jarzebski/Arduino-MPU6050 #4: The Wire.requestFrom does not need a Wire.beginTransmission, no Wire.endTransmission and no waiting.

Koepel, 2017-05-05:
In the file MPU6050.cpp the use of Wire.requestFrom can be improved.

The Wire.beginTransmission and Wire.endTransmission are only needed when writing data. They should not be used with Wire.requestFrom and can be removed. In two places there is Wire.beginTransmission, and in three places both the Wire.beginTransmission and Wire.endTransmission with the Wire.requestFrom. They can all be removed.

The Wire.requestFrom is a complete I2C transaction on its own. It stores the received data in a buffer in the Wire library and returns when the I2C transaction completely has finished. It is therefor not needed to wait for something. In four places there is a while-loop after Wire.requestFrom. Those while-loops can be removed.

J3r3Xy, 2017-12-14:
I agree with Koepel, the loops also cause the Arduino to hang if the MPU6050 disconnects (either because of cable or power failure)

Koepel, 2019-05-12:
If he can fix it, why can't you ?
https://github.com/ErcBunny/VTOL_osprey/issues/1