jarzebski/Arduino-MPU6050 #5: Multiple Sensors

Jobayori, 2017-07-30:
Can you add a method for adding multiple sensors to the Arduino's? I'm not sure how to connect one to 0x68 and 0x69 and call both in a program. Thanks

J3r3Xy, 2017-12-14:
To use two sensors you just need to define two instances of the class and specify the address in the begin() function, like this:
```
MPU6050 mpu68;
MPU6050 mpu69;
void setup()
{
    // ...
    // Syntax: MPU6050.begin(dps, range, address);
    mpu68.begin(MPU6050_SCALE_2000DPS, MPU6050_RANGE_2G, 0x68);
    mpu69.begin(MPU6050_SCALE_2000DPS, MPU6050_RANGE_2G, 0x69);
    // ...
}
```

pierinagalveza, 2020-06-23:
Hi!

Can you add a method for adding multiple MPU6050 to Arduino? I mean 3 or more.

Thanks a lot