jarzebski/Arduino-MPU6050 #7: Arduino ide for esp8266 use MPU6050

ehitil, 2017-10-22:
I use  arduino-MPU6050-master library , No matter how I flip it my serial output is positive and I can't get negative numbers。

like this:

 Xnorm = 0.08 Ynorm = 38.57 Znorm = 7.81
 Xraw = 84.00 Yraw = 64588.00 Zraw = 13068.00
 Xnorm = 0.14 Ynorm = 38.64 Znorm = 7.82
 Xraw = 232.00 Yraw = 64724.00 Zraw = 12860.00
 Xnorm = 0.18 Ynorm = 38.74 Znorm = 7.77
 Xraw = 36.00 Yraw = 64540.00 Zraw = 13124.00
 Xnorm = 0.03 Ynorm = 38.63 Znorm = 7.85
 Xraw = 64744.00 Yraw = 65072.00 Zraw = 10928.00
 Xnorm = 38.79 Ynorm = 38.95 Znorm = 6.97
 Xraw = 65244.00 Yraw = 65080.00 Zraw = 13208.00
 Xnorm = 39.00 Ynorm = 38.86 Znorm = 7.78
 Xraw = 65284.00 Yraw = 64800.00 Zraw = 13284.00
 Xnorm = 38.98 Ynorm = 38.78 Znorm = 7.91
 Xraw = 65116.00 Yraw = 1248.00 Zraw = 12456.00
 Xnorm = 39.03 Ynorm = 0.20 Znorm = 7.50
 Xraw = 64716.00 Yraw = 64716.00 Zraw = 13124.00
 Xnorm = 38.69 Ynorm = 38.82 Znorm = 7.83

ehitil, 2017-10-22:
i use +-2g but "norm" data more than 2,i can not understand why？

ehitil, 2017-10-22:
my code:

/*
    MPU6050 Triple Axis Gyroscope & Accelerometer. Simple Accelerometer Example.
    Read more: http://www.jarzebski.pl/arduino/czujniki-i-sensory/3-osiowy-zyroskop-i-akcelerometr-mpu6050.html
    GIT: https://github.com/jarzebski/Arduino-MPU6050
    Web: http://www.jarzebski.pl
    (c) 2014 by Korneliusz Jarzebski
*/

#include <Wire.h>
#include <MPU6050.h>

MPU6050 mpu;

void setup() 
{
  Serial.begin(115200);

  Serial.println("Initialize MPU6050");

  while(!mpu.begin(MPU6050_SCALE_2000DPS, MPU6050_RANGE_2G))
  {
    Serial.println("Could not find a valid MPU6050 sensor, check wiring!");
    delay(500);
  }

  // If you want, you can set accelerometer offsets
  // mpu.setAccelOffsetX();
  // mpu.setAccelOffsetY();
  // mpu.setAccelOffsetZ();
  
  checkSettings();
}

void checkSettings()
{
  Serial.println();
  
  Serial.print(" * Sleep Mode:            ");
  Serial.println(mpu.getSleepEnabled() ? "Enabled" : "Disabled");
  
  Serial.print(" * Clock Source:          ");
  switch(mpu.getClockSource())
  {
    case MPU6050_CLOCK_KEEP_RESET:     Serial.println("Stops the clock and keeps the timing generator in reset"); break;
    case MPU6050_CLOCK_EXTERNAL_19MHZ: Serial.println("PLL with external 19.2MHz reference"); break;
    case MPU6050_CLOCK_EXTERNAL_32KHZ: Serial.println("PLL with external 32.768kHz reference"); break;
    case MPU6050_CLOCK_PLL_ZGYRO:      Serial.println("PLL with Z axis gyroscope reference"); break;
    case MPU6050_CLOCK_PLL_YGYRO:      Serial.println("PLL with Y axis gyroscope reference"); break;
    case MPU6050_CLOCK_PLL_XGYRO:      Serial.println("PLL with X axis gyroscope reference"); break;
    case MPU6050_CLOCK_INTERNAL_8MHZ:  Serial.println("Internal 8MHz oscillator"); break;
  }
  
  Serial.print(" * Accelerometer:         ");
  switch(mpu.getRange())
  {
    case MPU6050_RANGE_16G:            Serial.println("+/- 16 g"); break;
    case MPU6050_RANGE_8G:             Serial.println("+/- 8 g"); break;
    case MPU6050_RANGE_4G:             Serial.println("+/- 4 g"); break;
    case MPU6050_RANGE_2G:             Serial.println("+/- 2 g"); break;
  }  

  Serial.print(" * Accelerometer offsets: ");
  Serial.print(mpu.getAccelOffsetX());
  Serial.print(" / ");
  Serial.print(mpu.getAccelOffsetY());
  Serial.print(" / ");
  Serial.println(mpu.getAccelOffsetZ());
  
  Serial.println();
}

void loop()
{
  Vector rawAccel = mpu.readRawAccel();
  Vector normAccel = mpu.readNormalizeAccel();

  Serial.print(" Xraw = ");
  Serial.print(rawAccel.XAxis);
  Serial.print(" Yraw = ");
  Serial.print(rawAccel.YAxis);
  Serial.print(" Zraw = ");

  Serial.println(rawAccel.ZAxis);
  Serial.print(" Xnorm = ");
  Serial.print(normAccel.XAxis);
  Serial.print(" Ynorm = ");
  Serial.print(normAccel.YAxis);
  Serial.print(" Znorm = ");
  Serial.println(normAccel.ZAxis);
  
  delay(1000);
}

T94T, 2018-03-07:
Data types do have different memory sizes assigned when using ESP platform.
Quick fix:
`Serial.print((int16_t)rawAccel.XAxis);`
This needs to be fixed in this repository.

d21d3q, 2018-05-15:
moreover `rawAccel.XAxis` is represented in memory as float, but [here](https://github.com/jarzebski/Arduino-MPU6050/blob/940449cfe16caa95fd899c35673e313a05ae3733/MPU6050.cpp#L376) data from registers are being thrown to float as if it were `int16_t`...
What king of conversion is that? 
(I was testing it on ESP and I had to cast it to `int16_t`. Does it work as expected on avr?)

fabiuz7, 2018-05-28:
Yeah found the same problem. In original Arduino environment this makes sense, the int were 16bit, than the Vector class was intended for general calculation..