jarzebski/Arduino-MPU6050 #13: get  gravity free  acceleration or linear acceleration

avsingh999, 2018-06-08:


AhmadSalmanKhan, 2019-07-19:
MPu6050 provides angular acceleration but I doubt it provides linear accelereation. You can check this from the fact that an accleerometer (linear one) should provide the acceleration when it's moving but give value of 0 when it's stationary or not moving. But that's not the case with mpu-6050 as it still continues to provide values even when stationary .

I've also checked multiple online sources but can't provide anything that shows mpu-6050 providing linear acceleration.

ben55j, 2019-11-03:
Hello, I got the same issue. I need to get a Linear acceleration and not an angular one. Mup-6050 seems to only provide angular one. How did you managed? Did you get another sensor?

AhmadSalmanKhan, 2019-11-06:
Hi,

Yes MPU6050 could potentially provide linear acceleration but by additional
processing of data using the original invensense software called motion
driver.
But the size of that was much greater than what our microcontroller could
support so we used ADXL335.

Good Luck and Happy coding.
Regards,

closed 2023-05-18 by jarzebski