jarzebski/Arduino-MPU6050 #25: Wiring failure

brrnd99, 2020-10-22:
Hi, im having issue on the result shown on serial monitor, my connection is as below:
VCC-----3.3V ( tried connected to 5V but getting same result)
GND-----GND
SCL-------A5
SDA------A4
AD0------GND ( tried to disconnect this but getting same result)

This is what i get on serial monitor:
[image]
My aim is to get pitch, roll and yaw angle( focusing more to get accurate yaw).  Any help for this problem is much appreciated:)

Janker16, 2021-02-21:
Having the same problem

Janker16, 2021-02-21:
Did you #brrnd99 fix it ?

umairdadaa, 2024-04-18:
Sorry for the late reply.

I had faced the same issue. Mine didn't work on A5 / A4. I am using direct SCA / SCL

- VCC: 5V
- GND: GND
- SCL: A5 / SCA
- SDA: A4 / SCL

Please also try to use cables that work. I am using the Arduino Mega board which has dedicated ports for SCA and SCL, so I'd recommend trying that.